vedderb/vesc_tool #7: Segfault when connecting to VESC

fabolhak, 2017-10-06:
Hello, 
thank you for the great tool. It looks pretty cool :). However I'm having difficulties to connect to my VESC, because the tool crashes, when pressing the "Connect" button.

I fired up the debugger and the tool crashes on this line:
https://github.com/vedderb/vesc_tool/blob/master/vescinterface.cpp#L312

It seems that the `ser` object is invalid. Looking in `listSerialPorts()` I figured that this may be due to the returned `res` stack-object which is destroyed after leaving the function:
https://github.com/vedderb/vesc_tool/blob/master/vescinterface.cpp#L396

I guess the a solution would be to pass a reference to an object `ser` which is filled inside the function and the returned.

fabolhak, 2017-10-06:
I got it compiling correctly by adding `CONFIG += c++11` to the vesc_tool.pro file :)

rewolff, 2017-11-08:
It compiles without any additions. However, the warning: 
./```
vescinterface.h:132:27: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool mSendCanBefore = false;
```
indicates to me that it is ignoring the intializer! As, in this case, it is likely that the "false" turns into zero anyway, chances are that it will still work. But with hundreds of these warnings, it is entirely possible that a non-zero initializer is required somewhere.... 

In fact, the compiled binary locks up quite quickly in my case, Not sure what the cause is, but I'm going to try adding the C++11....

closed 2019-02-20 by vedderb