vedderb/vesc_tool #18: build_lin tries to rm stuff around root if builds fail

leshy, 2018-06-20:
wow this was terrifying :D

if make fails, build/lin folder is never created and
```sh
cd build/lin
```
fails, but
```sh
cd ../..
```
passes, until it reaches root.
it will try to zip your whole hard drive, which is fine, but after it will run

```sh
ls | grep -v '\.zip$' | xargs rm
```

which will remove files @ / that you have access to, which are probably none, and will fail with a scary
```
rm: cannot remove 'bin': Permission denied                        
rm: cannot remove 'blocks': Is a directory                  
rm: cannot remove 'boot': Is a directory                              
rm: cannot remove 'dev': Is a directory        
rm: cannot remove 'etc': Is a directory                                                                           
rm: cannot remove 'frontend': Is a directory  
rm: cannot remove 'home': Is a directory        
rm: cannot remove 'keybase': Is a directory         
rm: cannot remove 'lib': Permission denied        
rm: cannot remove 'lib64': Permission denied    
rm: cannot remove 'lost+found': Is a directory                
rm: cannot remove 'mnt': Is a directory         
rm: cannot remove 'opt': Is a directory                       
rm: cannot remove 'proc': Is a directory                                                                                                                                             
rm: cannot remove 'root': Is a directory                                                                                                                      
rm: cannot remove 'run': Is a directory                                                                                                                     
rm: cannot remove 'sbin': Permission denied                                                                                                                      
rm: cannot remove 'server': Is a directory                                                                                                                     
rm: cannot remove 'srv': Is a directory                                                                                                                                     
rm: cannot remove 'sys': Is a directory                                                                                                                                
rm: cannot remove 'tmp': Is a directory                                                                                                                                              
rm: cannot remove 'usr': Is a directory                                                                                                                       
rm: cannot remove 'var': Is a directory                                                                                                                     
```

rpasichnyk, 2018-10-03:
Add
`set -e`
in the beginning of the build script

vedderb, 2019-02-20:
set -e is added, hopefully that solves the problem.

closed 2019-02-20 by vedderb