vedderb/vesc_tool #23: How to compile from source?

ElectricRCAircraftGuy, 2018-09-04:
[bldc-tool](https://github.com/vedderb/bldc-tool) has some nice instructions on how to compile from source. Can you add instructions to compile `vesc_tool` from source too?

rpasichnyk, 2018-09-05:
Download Qt, open Qt Creator, open vesc_tool.pro, hit build

subodh-malgonde, 2018-09-17:
For Ubuntu, the same instructions as those for [bldc-tool](https://github.com/vedderb/bldc-tool) worked for me, with a minor modifications:
1. I chose not to install `qtcreator` and `qt-sdk`. I chose to install `qt5-default` instead, as recommended over [here](https://github.com/RacecarJ/installBLDCTool/blob/master/installBLDCToolHost.sh). This is not specifically for the vesc_tool, it works for the bldc_tool as well.
 
2. In addition to `libudev, libqt5serialport5`, I had to install `qtconnectivity5`.

Full instructions:

1. `sudo apt-get install qt5-default libudev-dev libqt5serialport5-dev qtconnectivity5-dev -y`
2. `cd vesc_tool`
3. `qmake -qt=qt5`
4. `make clean && make`
5.  Allow for serial access without using sudo: `sudo adduser $USER dialout`
6. Restart for access changes to take effect `sudo reboot now`
7. Start VESC Tool from inside of the built repo `./vesc_tool_095` (I am using version [0.95](https://github.com/rpasichnyk/vesc_tool/releases/tag/v0.95-fw3.40) of the tool)

subodh-malgonde, 2018-09-17:
Also precompiled Mac executable is available at this repository https://github.com/rpasichnyk/vesc_tool/releases

vedderb, 2019-02-20:
there are also some instructions here:
https://vesc-project.com/node/310

closed 2019-02-20 by vedderb