vedderb/vesc_tool #24: build_linux deleted most of my home and tried to delete folders from /*

Alkeryn, 2018-09-22:
Luckily i have backups so i don't think i loosed anything but still
anyway here is the ouptut it made
```
❯ ./build_lin
Info: creating stash file /home/alkeryn/tmp/vesc_tool/.qmake.stash
Project ERROR: Unknown module(s) in QT: quickcontrols2 serialport
make: *** Aucune règle pour fabriquer la cible « clean ». Arrêt.
make: *** Pas de cible spécifiée et aucun makefile n'a été trouvé. Arrêt.
./build_lin: ligne 31 : cd: build/lin: Aucun fichier ou dossier de ce type
  adding: android/ (stored 0%)
  adding: application/ (stored 0%)
  adding: bleuart.cpp (deflated 73%)
  adding: bleuart.h (deflated 59%)
  adding: build_all (deflated 66%)
  adding: build_android (deflated 67%)
  adding: build_cp_fw (deflated 27%)
  adding: build_lin (deflated 66%)
  adding: build_lin_original_only (deflated 23%)
  adding: build_win (deflated 71%)
  adding: build_win_original_only (deflated 28%)
  adding: commands.cpp (deflated 79%)
  adding: commands.h (deflated 64%)
  adding: configparam.cpp (deflated 58%)
  adding: configparam.h (deflated 53%)
  adding: configparams.cpp (deflated 87%)
  adding: configparams.h (deflated 74%)
  adding: datatypes.h (deflated 68%)
  adding: digitalfiltering.cpp (deflated 69%)
  adding: digitalfiltering.h (deflated 52%)
  adding: main.cpp (deflated 62%)
  adding: mainwindow.cpp (deflated 79%)
  adding: mainwindow.h (deflated 69%)
  adding: mainwindow.ui (deflated 89%)
  adding: mobile/ (stored 0%)
  adding: packet.cpp (deflated 64%)
  adding: packet.h (deflated 48%)
  adding: pages/ (stored 0%)
  adding: parametereditor.cpp (deflated 78%)
  adding: parametereditor.h (deflated 59%)
  adding: parametereditor.ui (deflated 91%)
  adding: README.md (deflated 19%)
  adding: res/ (stored 0%)
  adding: res_bronze.qrc (deflated 43%)
  adding: res_free.qrc (deflated 44%)
  adding: res_fw_original.qrc (deflated 74%)
  adding: res_fw.qrc (deflated 85%)
  adding: res_gold.qrc (deflated 43%)
  adding: res_neutral.qrc (deflated 44%)
  adding: res_original.qrc (deflated 44%)
  adding: res_platinum.qrc (deflated 44%)
  adding: res.qrc (deflated 84%)
  adding: res_silver.qrc (deflated 44%)
  adding: setupwizardapp.cpp (deflated 82%)
  adding: setupwizardapp.h (deflated 79%)
  adding: setupwizardmotor.cpp (deflated 80%)
  adding: setupwizardmotor.h (deflated 78%)
  adding: startupwizard.cpp (deflated 76%)
  adding: startupwizard.h (deflated 64%)
  adding: utility.cpp (deflated 65%)
  adding: utility.h (deflated 50%)
  adding: vbytearray.cpp (deflated 73%)
  adding: vbytearray.h (deflated 58%)
  adding: vescinterface.cpp (deflated 76%)
  adding: vescinterface.h (deflated 65%)
  adding: vesc_tool.pro (deflated 67%)
  adding: widgets/ (stored 0%)
rm: impossible de supprimer 'android': est un dossier
rm: impossible de supprimer 'application': est un dossier
rm: impossible de supprimer 'mobile': est un dossier
rm: impossible de supprimer 'pages': est un dossier
[... 384 lines cut ...]
make: *** Aucune règle pour fabriquer la cible « clean ». Arrêt.
make: *** Pas de cible spécifiée et aucun makefile n'a été trouvé. Arrêt.
./build_lin: ligne 81 : cd: build/lin: Aucun fichier ou dossier de ce type
zip I/O error: Permission denied
zip error: Could not create output file (vesc_tool_free_linux.zip)
rm: impossible de supprimer 'bin': Permission non accordée
rm: impossible de supprimer 'boot': est un dossier
rm: impossible de supprimer 'dev': est un dossier
rm: impossible de supprimer 'etc': est un dossier
rm: impossible de supprimer 'home': est un dossier
rm: impossible de supprimer 'lib': Permission non accordée
rm: impossible de supprimer 'lib64': Permission non accordée
rm: impossible de supprimer 'mnt': est un dossier
rm: impossible de supprimer 'opt': est un dossier
rm: impossible de supprimer 'proc': est un dossier
rm: impossible de supprimer 'root': est un dossier
rm: impossible de supprimer 'run': est un dossier
rm: impossible de supprimer 'sbin': Permission non accordée
rm: impossible de supprimer 'srv': est un dossier
rm: impossible de supprimer 'sys': est un dossier
rm: impossible de supprimer 'tmp': est un dossier
rm: impossible de supprimer 'usr': est un dossier
rm: impossible de supprimer 'var': est un dossier
```
Also yeah output is french

mihlit, 2018-10-02:
I can confirm this, this is because it tries to 'cd' to directory that would be created during build, but if build fails because of problem with a dependency, 'cd' fails and it continues with other commands that include 'rm'

from build script:

cd build/lin
zip vesc_tool_original_linux.zip `ls | grep -v '\.zip$'`
ls | grep -v '\.zip$' | xargs rm

solution would be to either:
a)
cd build/lin | exit 1

or
b)
cd build/lin
zip vesc_tool_original_linux.zip `ls | grep -v '\.zip$'`
cd ../..
ls build/lin | grep -v '\.zip$' | xargs rm

mihlit, 2018-10-02:
I think this is a duplicate of issue #18

vedderb, 2019-02-20:
I have added set -e to the build script, hopefully that resolves the issue.

closed 2019-02-20 by vedderb