vedderb/vesc_tool #36: Dynamic arrays supported by GCC compiler but not for ms compiler in digitalfiltering.cpp

SigmaOrionis, 2019-06-17:
digitalfiltering.cpp 

	std::vector<double> imag = std::vector<double>( taps );
	std::vector<double> filter_vector = std::vector<double>( taps );
	//double* imag[ taps ] = new double[ taps ];
	//double* filter_vector[ taps ] = new double[ taps ];

GCC allows dynamic arrays but Visual Studio compiler does not. You can't compile VESC tool on VS. It would be better to use for example std::vector for dynamic array then it would compile.
If it is multi-platform then make it multi-platform...

Teslafly, 2019-06-18:
You can install gcc on windows and set vs to use it. Since you are cross compiling anyway this really has nothing to do with multi platform support.

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Looks like 4.9.3.2015q3 is used

SigmaOrionis, 2019-06-19:
You are right.

closed 2019-06-19 by SigmaOrionis