vedderb/vesc_tool #40: NRF5x firmware source? NRF52840 SKB501 module

Niuva, 2019-07-13:
Seems like the firmware for NRF52840 uses a pin (PO.11) that is not present on Skylab SKB501 modules.
Could we perhaps get our hands on the source of the firmware? I'd love to tinker with it.

Thanks! <3

Teslafly, 2019-07-16:
Is this the fw source you are looking for?
https://github.com/vedderb/nrf51_vesc
Btw you may have to experiment with the different sdk versions. I don't think vedder specifies which version works and I forget which sdk kinda worked for me (16k ram builds failed) when I tried compiling.

Niuva, 2019-07-16:
I'm afraid that's only for the NRF51 modules. I'm looking for the NRF52 code.
Benjamin has released a few pre-compiled ones, but no NRF52 source. Only the 51.
https://github.com/vedderb/vesc_tool/tree/master/res/other_fw

vedderb, 2019-07-17:
I have a version with the NRF52 SDK as well, but haven't had an opportunity to clean it up and upload it yet. I will look into it soon and leave this issue open until then.

vedderb, 2019-07-19:
Here is the NRF52 code:
https://github.com/vedderb/nrf52_vesc
There is some information in the readme, hopefully it is enough to get started with it.

closed 2019-07-19 by vedderb