vedderb/vesc_tool #41: New to github plus all VESC tool online is down.

retalinate, 2019-07-15:
I've gone to every website to get this VESC tool because I think it will help me get my electric longboard running but I can't seem to find anything other than this source code. I have it extracted but have no idea what to do from here. Any help would be much appreciated.

In addition, I've tried using the BLDC tool and it said my VESC firmware is too old. I have a 4.12 VESC and 2.18 BLDC tool. I tried turning on the motor without adjusting anything on a online tool and I'm scared I might be damaging the motor by doing so.

AntumArk, 2019-07-15:
Yep site is down. @retalinate you could download QT creator and run   project for start

vedderb, 2019-07-15:
My server has died, and I'm away on vacation. I managed to set up a build on my laptop and create a temporary website that http://vesc-project.com should redirect to. You might have to clear your DNS cache for it to work. Otherwise you can go here: http://vedder.se/vescproject_tmp/

closed 2019-07-15 by vedderb