vedderb/vesc_tool #58: Please provide documentation for building support to a FW version

Peemouse, 2020-01-14:
Hi Benjamin,

The new VESC Tool 2.0 is a huge step in the history of the project !
Thanks for this amazing work.

Could you provide a quick HOW-TO or tutorial for letting us build and contribute to add support to older FW than 3.60 ?

I guess that is a tedious work to do and you may wish focus on more interesting points.

Thanks in advance !

Clément

vedderb, 2020-01-15:
I'm glad you liked the update!

It is fairly simple to add support for older firmwares, but a bit tedious as you said. The steps are.

1. Find parameters_mcconf.xml, mcconf_appconf.xml and info.xml for the firmware version and put them in a directory under res/conf with the firmware name. You can find the files by going through the git history of VESC Tool. Info.xml has a field that says which firmware version is supported.

2. Add the grouping information at the end, like this:
https://github.com/vedderb/vesc_tool/blob/master/res/config/3.66/parameters_mcconf.xml#L3118
you can copy the information from a newer firmware and then update it according to the parameters that are there. SerInfo shows which parameters are present, and a diff tool can help a bit.

3. Add the configs to res_config.qrc
https://github.com/vedderb/vesc_tool/blob/master/res_config.qrc

This should work back to FW 3.55 as that is where I changed how signatures are generated. Supporting even older firmware is also possible, but then I have to look into confnegerator and have rules on signature calculation based on the firmware version.

Peemouse, 2020-01-15:
Thanks a lot Ben !
I try to catch some free time to contribute.

> This should work back to FW 3.55 

This is also the info I wanted to know. You read into my mind :)

Peemouse, 2020-01-29:
Hi Ben,

Congratulations for the FW4.00 ! This is art work !
I'm impressed even more at each new release !

I saw that signatures have changed with this new FW.
Based on your previous comment :

> Supporting even older firmware is also possible, but then I have to look into confnegerator and have rules on signature calculation based on the firmware version

Does that mean that it break the backward compatibility ?

vedderb, 2020-01-29:
Thanks :-)

It should still work. The signatures are different between all firmwares, but generating them gives the correct result if the xml files are correct. The thing that changed after 3.55 is how the signature is generated, but that has not changed since. Changing the combination of parameters will give a new unique signature, so changing the parameters between firmwares changes the signature between them.

The reason for the signature is to avoid uploading a configuration that has an incompatible byte order, which can mess up the settings quite badly. This is what happened with the metr app a while ago, which is why I introduced the signatures.

closed 2020-01-29 by vedderb

Peemouse, 2020-01-29:
Got it.
Thanks for the explanation.