vedderb/vesc_tool #73: Cannot find running Bluez, please check the Bluez installation

jxltom, 2020-04-02:
I'm using Ubuntu 16.04. This causes around 30 seconds to boot up the VESC and shows this error ```Cannot find running Bluez, please check the Bluez installation```.

It looks like ```new QBluetoothDeviceDiscoveryAgent``` is causing this issue.

vedderb, 2020-04-02:
I'm aware of this, but haven't found a good solution yet. Please let me know if you find one.

jxltom, 2020-04-03:
It might relate to qt's library instead of vesc, but i didn't dig it down.

I will update here if I found any solutions.

Jfriesen222, 2021-10-04:
This issue has been more or less addressed and we can probably close it.  There is a bug in the Qt BLE implementation that causes the creation of the BLE service controller to hang for 30 seconds on some platforms when bluetooth is not present. As a workaround the bluetooth controller is not instantiated until the bluetooth scan button is clicked which causes the tool to then hang for 30 seconds or so if no bluetooth is present. It used to be instantiated on launch causing the hang every time the tool was opened. 

Unfortunately due to the multi-platform nature of the tool doing much more is difficult but I don't think many people without bluetooth in the computer will be using the bluetooth scan button anyways and it doesn't crash or anything but just wait for a 30 second time-out.

closed 2021-10-05 by vedderb