vedderb/vesc_tool #75: VESC Tool crashing on windows

hichemhlm, 2020-04-11:
The Vesc tool is now crashing on windows, i have tested 2.02 and the 2.03. 
It was working fine and i think after a windows update(not sur) it is not working any more and crashing with access memory violation(debug with Visual studio). 
I have already tried to remove updates but no result

albfan, 2020-11-25:
See it needs specific Qt version. At least in linux is QT 5.9, 5.12 or 5.13. You can download precompiled versions on vesc tool site to check if it really works or not.

Teslafly, 2022-12-06:
This is probably fixed or a non issue by now. But also falls under the subset of not having great developer documentation leading to building issues.
https://github.com/vedderb/vesc_tool/issues/92

Can this be closed?

closed 2022-12-06 by vedderb