vedderb/vesc_tool #76: Android app shows duplicate of one VESC UUID when backing up and restoring.

skatardude10, 2020-04-16:
https://forum.esk8.news/t/vesc-fw-5-0-beta-testers-wanted/23342/238?u=skatardude10

As in the title. Connected to master, when backing up or restoring, the notification saying it succeeded shows two of the same UUID listed. CAN Forward to the second VESC and it shows duplicates of the other UUID on the pop up at the end.

All unique settings seem to apply just fine for each individual VESC though.

vedderb, 2020-04-17:
I just gave it a try, and I saw different UUIDs both in the backup and restore dialog. Did you happen to use a focbox unity? It would probably show the same UUID twice. I fixed that by adding 1 to the UUID of motor 2.

skatardude10, 2020-04-17:
It is on a dual setup via CAN on Neobox V6, master ID 100 and Slave 98, not unity. Strange, the other day it showed a different UUID when connecting via CAN and performing the process. hw_status connected to either master or slave show they have the same UUID now though.

Curious if this is a hardware problem.

[image]
[image]

skatardude10, 2020-04-17:
Disregard, missed a single number, my bad.

closed 2020-04-17 by skatardude10