vedderb/vesc_tool #79: Adding serialbus to Qt seems to break the serial port on static builds

nitrousnrg, 2020-04-26:
This line says that serialbus breaks serialport:
https://github.com/vedderb/vesc_tool/blob/master/vesc_tool.pro#L44

I tried building VESC Tool 2.04 statically for linux (./build_lin_original_only) and both USB serial and canbus interfaces work for me.
I also compiled VESC Tool for ARM and tested this on a raspberry pi and both serial and canbus worked for me using ./build_lin_original_only (the binary needs some dependencies to run there)

So I couldn't reproduce the issue... should I test it on windows machines?

vedderb, 2020-04-27:
It works when compiling normally, but when I use my static build on Ubuntu 16.04 and run it on different machines, the serial port does not work on some of them. I will look into is more soon as our large battery is about to arrive, and then I will start testing with your controller. All parts are here on my desk waiting for the battery.

p0wdrdotcom, 2022-12-07:
https://github.com/vedderb/vesc_tool/pull/279