vedderb/vesc_tool #89: VESC Tool incorrectly reports firmware mismatch during Input Setup Wizard

bovine, 2020-07-05:
I have VESC Tool 2.06 on Windows 10 and attempting to run the Input Setup Wizard causes the following error even though both of my ESCs are running the same version:

> "Your VESC (or one of the VESCs on the CAN-bus) has old firmware, and needs to be updated. After that, the motor configuration has to be done again."

I am connected to the primary VESC over USB, and the secondary VESC is connected to the primary over CAN bus.  Here is the output from the debug console and you can see both report as having firmware version 5.1:
```
2020-07-05 18:18:52: Status: MC configuration updated
2020-07-05 18:18:52: Status: App configuration updated
2020-07-05 18:18:55: Status: VESC Firmware Version 5.1, Hardware: 410, UUID: 43 00 35 00 10 50 4D 4D 4E 39 37 20
2020-07-05 18:18:56: Status: VESC Firmware Version 5.1, Hardware: 410, UUID: 35 00 1F 00 06 50 4D 4D 4E 39 37 20
2020-07-05 18:18:59: Status: MC configuration updated
2020-07-05 18:18:59: Status: App configuration updated
...
2020-07-05 18:24:04: Status: App configuration updated
2020-07-05 18:24:04: Status: APPCONF Write OK
2020-07-05 18:24:06: WARNING (:0 ): Appconf not received
2020-07-05 18:24:06: Status: App configuration updated
2020-07-05 18:29:07: Status: App configuration updated
```

The WARNING message appears right before the moment the firmware mismatch error occurs, so maybe there is a timeout occurring during the wizard and causing the `isLimitedMode()` codepath? https://github.com/vedderb/vesc_tool/blob/master/setupwizardapp.cpp#L104

It would be nice if there was a checkbox that could override the firmware version check and allow the wizard to continue.

bovine, 2020-07-05:
As a workaround, temporarily disconnecting the CAN bus wire between the two VESCs seems to allow the wizard to proceed.

StrikeLines, 2021-03-16:
Exact same issue here. Dual vesc6 connected by CAN bus.  Firmware incorrectly reports as mismatch when I try to run the input control wizard. This always occurs while I'm trying to run the input wizard, not during other parts of the configuration process. 

Waiting a few moments and trying again sometimes results in the firmware being correctly detected and the wizard completing successfully.

rusins, 2023-05-15:
Experiencing the same issue with vesc tool 6.02. Kept getting this error with 2 Trampa vesc6 mk3 connected through can with metr can in the middle. Interestingly, first switching to the vesc over CAN in vesc tool, and then pressing the input wizzard button, worked :shrug: