vedderb/vesc_tool #90: Please create version tags and github "releases"

bovine, 2020-07-06:
Furthering the branching effort done in issue #47 to improve the ability for other developers to differentiate production code, it would be awesome if you could begin creating git tags for each version number that you publish. You can optionally choose to also create a Github "release" for each git tag that allows you to summarize the release notes or major feature changes, in human-readable text. 

Version tags and releases are a key step to making it easier for people that need to track down what VESC Tool versions a new change first became available in, or might have been affected by a previously addressed bug. It also makes it easy to determine what changes are new since a prior version.

TechAUmNu, 2021-08-19:
The releases are particularly important when the main https://vesc-project.com/ website is down and there is no way to download. 
Such as right now.

b264, 2022-06-15:
I second this request.

albfan, 2025-02-01:
At this point, looks like commits are use to do releases, and branches release_6_02 release_6_05 are use for bugfixing.

probably on top of each release_x_xx tags like 6.05.00, 6.05.01 can be use. That would help to do releases on flatpak.