vedderb/vesc_tool #92: Add compiling documentation

Teslafly, 2020-08-20:
The compiling documentation on http://vedder.se/2015/01/vesc-open-source-esc/ is out of date.
the compiling information should really live in vcs .md files to be a single source living document.

The readme.md or a separate compiling.md should give step by step instructions on how to set up an environment and compile vesc tool from a clean system. Preferably working on at least ubuntu 20.04 and windows (possibly through wsl)

I will take a stab at creating this documentation as I figure it out.

albfan, 2021-01-11:
In general describe how to build a project is not a task for project itself, as the steps for each OS are different.

Just to put an example, these are the build steps for arch linux

https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=vesc_tool-git

Apart from having a buildsystem (Qt here) there's few a project can do to document how to build it properly anywhere. It depends on where you want to build it.

Probably a wiki can help, but it is going to be outdated from time to time, if nobody cares of it.

I suggest to close this issue as there's no real action to do on it.

vedderb, 2021-01-15:
I think some information in the readme file would be nice. It is fairly simple to compile though, a default installation of Qt is enough. There is a list of apt-packages in vesc_tool.pro to install everything needed on ubuntu and raspbian (probably not the best place for it) that can be used to get started.

albfan, 2021-01-16:
Build on arch linux from source lead to a segfault, so not really sure. Probably point to existing scripts `build_*` would be enough

Teslafly, 2023-01-04:
Need to add a note about cloning/copying firmware files into vesc tool to avoid `File does not exist 'res/firmwares/res_fw.qrc'` errors. Initial setup instructions should include this step.

https://discord.com/channels/904830990319485030/918975003700121671/1059635258292715541

albfan, 2025-02-01:
Possible solutions to close this:

- BUILDING.md
- Open a wiki
- Quick link to a forum article

but in general this looks out of scope