vedderb/vesc_tool #108: Mini FSESC4.20 with two hall sensors as throttle (ADC) and brake (ADC2), brake signal polutes throttle

albfan, 2020-12-21:
Using Mini FSESC4.20 with two hall sensors as throttle (ADC) and brake (ADC2), why brake signal polutes throttle?

"Current No Reverse Brake ADC2" is even worst as suppose I have a center throttle and I start backwards!

https://user-images.githubusercontent.com/220968/102763865-8fa5a200-437a-11eb-8261-7682f6808816.mp4

I'm using VESC tool 2.06 free on linux (downloaded from your website) and upload last firmware on my VESC.

**NOTE:** I probably can rebuild tool if there's anything I can try

Teslafly, 2020-12-21:
If a problem exists with this, it would be on the vesc firmware (bldc) side. 

There shouldn't be any logic in vesc that would do this, and I have seen it as a result of a damaged stm32. Generally by overvoltage on one of the pins.

What are you using to drive Tha adc/adc2 pins respectively? Hall throttles? Connected to what voltages? 

That adc2 is reaching 3.3v is concerning that you may have damaged that throttle input. No adc inputs should go above 3.3v. If you are reaching 3.3v you are likely going above.

Teslafly, 2020-12-21:
I just tested it with vesc tool 2.06 and a hw 4.12 vesc. Both adc1 and adc2 inputs are completely seperate.
This rules out a firmware problem, and the most likely cause is a damaged vesc/stm32 or bad wiring. Check your adc1/2 input voltages. 

The power for all throttle inputs should be 3.3v. Do not use 5v to power hall throttles or potentiometers.

albfan, 2020-12-21:
Opps, my bad. Let's rewire it. I suspect that seing the max voltage.

Thanks for confirming!

closed 2020-12-21 by albfan