vedderb/vesc_tool #109: Please add a beta version number or similar

surfdado, 2021-01-06:
There've been close to 10 different beta releases of the VESC Tool since the 5.1/2.06 release. Unfortunately the version number has always been the same. 

Please add an additional revision/beta-version/git-commit or other ID that gets displayed in the About dialog of the VESC tool.

Thank you!

vedderb, 2021-01-11:
I agree, will add that to the next dev version. Leaving this issue open until then.

closed 2021-01-15 by vedderb

surfdado, 2021-05-13:
I don't know if you can see this comment given that the issue is closed now, but is it just me or is the test version in the about dialog always 1 ? It would be nice if it instead matched the beta number in the firmware...

commit ff92453b (2021-01-15): Added test version information to about dialog. Fixes #109