vedderb/vesc_tool #112: Distribute through flatpak

albfan, 2021-01-11:
To ease distribution on *nix environments I have created https://github.com/flathub/flathub/pull/2043 so VESC tool can be distributed through flatpak.

Let me know if you're interested on that kind of release to polish icon, metadata and do final submit.

flathub allows beta version if you need feedback for unrelease code.

I'm not sure about the options to release Platinum, Gold, Silver, Bronze through an app store that allows payments: More info here

https://github.com/flathub/flathub/issues/680

albfan, 2021-01-11:
[image]

albfan, 2021-01-13:
Build is ready on flathub, you can try with:

    flatpak install --user https://dl.flathub.org/build-repo/35270/com.github.VescTool.flatpakref

a first benefit is that tool honors theme, here dark version used on my system:

[image]

@vedderb Let me know if you approve this, so we can merge the app in flathub.

vedderb, 2021-01-13:
Not going via the vesc-project site misses an opportunity to give a donation, which is where 95% of the donations come from. That being said, people using a free OS is something I want to support, so it is nice to give them some advantages in the form of an easy flatpak install :-)

The one thing that worries me is that the built-in firmwares might be the wrong version if the build is pulled between when I update vesc tool and when I push the firmwares resource files. Is there some way for me to trigger which commit should be used for the next build, so that the built-in firmwares always are in sync correctly?

albfan, 2021-01-13:
Absolutely. You can bind to a tag, commit or build from tar if you prefer.

There're some work about paid apps on flatpak and site will link the donations in its metadata.

Just tell me where 2.06 version commit is and as soon as you release 2.07 or 3.00 we can launch a new version.

albfan, 2021-01-13:
I found this possible commits for versions (looking at VT_VERSION

```
* 520bd98 (v3.00) Added BMS commands, bumped up version, hide some pages when other HW is connected
* 2321785 (v2.07) Added FW5.02 files, added HW version check before upload all
* f992d39 (v2.06) Updated changelog and version number
* 4d46f37 (v2.05) Fixed PPM bug in previous release
* 6f4bc33 (v2.04) FW5 support, added NRF pair in VESC remote page, added FOC detect button without CAN
* 651fe50 (v2.03) Moved balance updates to FW4.02, changed default FOC time constant
* 9e442a1 (v2.02) Better autoconnect, increased maximum SWDPROG file size
* e8ddc0a (v2.01) Added HFI tab, disconnect after FW upload, FW3.67 support
* 7274a0f (v2.00) Version 2.00: many changes, see changelog
* 042240c (v1.29) Replaced miniLZO with lzokay and updated app generator
* edaf879 (v1.28) PTC motor thermistor support, FW 3.65
* 721e286 (v1.27) Experiment plot truncate option, LZO android fix
* 5746b4c (v1.26) Log analysis updates, FW compression support, better observer gain calculation
* e64cb94 (v1.25) Log analysis improvements, renamed nunchuk
* 2b8b978 (v1.24) Log analysis improvements
* b219681 (v1.23) Added more data to log analysis
* 531cf75 (v1.22) Use forground service on android during logging
* 2553dba (v1.21) Only hide GUI in hidden state, use opengl es2 for static linux build
* 5986063 (v1.20) Experimental wake lock and logging updates
* ce2cd05 (v1.18) V 1.19: Mobile parameter prefix fix and added smart reverse to PPM app
* 002a5c5 (v0.82) Added version number to executable, only show intro if it has changed
```

But looking in detail:

```
* 2321785 (v2.07) Added FW5.02 files, added HW version check before upload all
* f992d39 (v2.06) Updated changelog and version number
* bcc00e0 Made IMU calibration more compact
* 5c3a368 Only reset fwrx on can-id change while can fwd is connected
* cebb493 Moved test FW check to end of FwRx to avoid multiple popups due to the dialog blocking the function
* 4d46f37 (v2.05) Fixed PPM bug in previous release
* 87565a4 Rebuilt FWs and removed test flag
* ded5d60 Added 60_MK4 and 75_300_R3 firmwares
* 12a7106 Updated nrf51 fws again
```

Just let me know to release v2.06. For v2.07 or v3.00 I'm not sure as they are not available on your site. Would be great if you tag commits for that.

vedderb, 2021-01-14:
The plan with the master branch is that the latest commit always is stable, and that development is done on development branches that then are merged to master when they are ready. Sometimes new firmwares are added to master, but that does not change the version, it only updates the included firmwares. The only time the latest commit is not stable is the moment when I'm working on merging a development branch to master once it is tested and ready. This is only the case for an hour or so every two months or so, but it is good to be aware of that.

It should be safe to assume that if the latest commit in the master branch is more than two days old, it is the latest stable release. The version number can be found in in vesc_tool.pro.

albfan, 2021-01-14:
You have a workflow that works, just in case it helps, probably open a branch 2 where you can cherry pick changes will allow to create 2.07, 2.08, 2.09... etc.

Master can be on version 3, until you decide on a big change (version 4) where you create branch 3 and start to create tag 3.00, 3.01... this way you can even apply this fixes (like the problem with enums) in a 2.06.1 or similar (not always backport changes is possible)

So should I go with f992d39 as 2.06? Probably I will apply the enum fix on top of that so you can see what I mean.

vedderb, 2021-01-14:
Your fix is already in master, it was the last thing that made it for this release. You can go with https://github.com/vedderb/vesc_tool/commit/b0632c42e896175f9d731be5f146f0202a0fe17c and version 3.00. Next version will be 3.01, which will move to master when I feel that it is ready and tested.

At the moment I don't see any need for supporting old releases, as the new ones work with old firmwares too.

albfan, 2021-01-14:
Cool. Then from flatpak side everything is done:

Commit for version 3.00:
https://github.com/flathub/flathub/pull/2043/files#diff-80814107ffcce952be145518d76aae2eab08470dae27b935b4dc767147dd576aR18

Link to donations:

https://github.com/flathub/flathub/pull/2043/files#diff-da593c4d8a93b1b3f1f4c10a8df6457a4f093fc07f066a08ef0f21c3c7ffa6f3R36

I asked about how to connect the flatpak app with your current donation methods. Not really sure but probably that will need some kind of Rest API on your site (totally guessing at this time). Just ask that on last comment here: https://github.com/flathub/flathub/pull/2043#issuecomment-760496771

albfan, 2021-01-14:
spreading on social media: https://twitter.com/albfanjul/status/1349839463942397959

Seems after build different flatpak versions for a release (Platinum, Gold, Silver, Bronze, Free) they can be linked somehow to your website.

vedderb, 2021-01-15:
Nice, thanks for setting this up!

I gave it a quick try (my first time using flatpak), and noticed that the map in log analysis does not work. It does not seem to be able to download the openstreetmap tiles. Is there perhaps some missing permission in the flatpak build?

Also, will vesc tool show up on the flathub website or is there any extra action needed from me?

albfan, 2021-01-15:
Ah, as a sandbox, network permissions have to be explicity granted:

[image]

That should be fixed now

```
flatpak install --user org.kde.Platform//5.15
flatpak install --user https://dl.flathub.org/build-repo/35422/com.vesc_project.VescTool.flatpakref
flatpak run com.vesc_project.VescTool
```

albfan, 2021-01-16:
I just check an upload hardware firmware with flatpak app (3.00). Then I setup motor and input with it too. All work correctly.

If you want to distribute flatpakrefs on website I guess you will get all the goals. I cannot make anyone to explain me how paid apps should work on flatpak. I guess do something and wait for "someone is wrong in internet" would probably be the faster way to understand how it should work

albfan, 2021-02-10:
Done: https://flathub.org/apps/details/com.vesc_project.VescTool

Any mailing list or RSS to notify new versions? I will maintain the flatpak app.

closed 2021-02-10 by albfan

albfan, 2024-03-04:
In case of any issue with flatpak use https://github.com/flathub/com.vesc_project.VescTool/issues