vedderb/vesc_tool #113: Setup input wizard do not follow control type selected

albfan, 2021-01-16:
Tested on 2.06 and 3.00

After select "Current No Reverse Brake ADC2" on input wizard, next page shows "Current" as control type selected.

Only after select "Current No Reverse Brake ADC2" (or whatever type you're configuring) and write config it behaves as expected.

vedderb, 2021-01-18:
The intention of the preview is to show how each mode behaves, it is then up to the user to select the mode on the next page. I agree that this is a bit unintuitive, so it is probably better to select the mode based on the value in the adcmap box.

You can have a look here:
https://github.com/vedderb/vesc_tool/blob/master/setupwizardapp.cpp#L655

before writing the appconf, you can update the configuration with

`
mVesc->appConfig()->updateParamEnum("app_ppm_conf.ctrl_type", <control_type>);
`

the adcmap widget does not have a method to get the selected control type, but it is easy to add one.

albfan, 2021-01-19:
Yes makes sense. Probably with an empty selection that do not allow you to pass next page will be intuitive.

Anyway 99% of time people will use the selection on previous page (I guess)

Will look to make PR on this.

We can check if that is or not a better way to handle it.