vedderb/vesc_tool #116: Can't do datalogging on some Android phones

nitrousnrg, 2021-03-02:
In some android phones you can't do realtime data loggings.

This has been reported by a user that installed the app from google play:
[image]

In our team we can reproduce the same issue with a Note 9 phone.

Logging works on my samsung S8. The hardcoded "file:///sdcard/" is a bit confusing because the phone has no SD card, but I don't think its related to the logging issue.

I wish I could help fixing this, but don't even know where to start

vedderb, 2021-03-04:
It is not easy for me to test as I don't have a phone to reproduce this. What are the symptoms? Does the directory listing not work? Is it asking for file permissions properly? Does it help to grant the permission manually in the app settings?

nitrousnrg, 2021-03-04:
Yeah I don't have one of those with me either, but my mate 10000km away just sent me a couple of screenshots:

[image]

It does ask for file permissions.

[image]

vedderb, 2021-03-24:
I noticed the same problem on my phone today and made an attempt at fixing it:
https://github.com/vedderb/vesc_tool/commit/412e4672e8347f172e85c7ba8ebf93b753034a71
It does not fix it for me though, but I think this is how it is supposed to work. Can you give it a try? The apk here should be updated by now:
https://vesc-project.com/node/2859

nitrousnrg, 2021-03-24:
Got it! Forwarding to the team

vedderb, 2021-03-24:
I just got it working on my phone too, there was a mistake in my fix. Building a new version now, should be up any moment now.

nitrousnrg, 2021-04-06:
Fixed! I'm closing this now, thanks!

closed 2021-04-06 by nitrousnrg