vedderb/vesc_tool #120: QGamepadManager hai

BlackZcc, 2021-04-02:
QGamepadManager::instance()->gamepadName(g)

BlackZcc, 2021-04-02:
Hello 
I can't build the new version 0.95,as it debug "QGamepadManager has no method as gamepadName()" for this line---QGamepadManager::instance()->gamepadName(g) in pages/PageSettings.cpp in line 111 for return the device name on order to delete the error device;

 As I check the official QT class QGamepadManager  do not contain such method,the you old version 0.91 dose use this class.
Is that QGamepadManager class your own class or  I make some cofigurion error.

ravens2, 2022-01-12:
I'm having issues with the gamepad as well.

I'm running the project in QtCreator with qmake reduced version 6.22.2 (community/open-source). I get an error when I try to compile the code that says:
`Project ERROR: Unknown module(s) in QT: gamepad`

Having a look at the files I can see that for android applications the gamepad is called. Doing a search for Qt's forums for a similar error/fix I came across [this post](https://bugreports.qt.io/browse/QTBUG-61553?focusedCommentId=371230&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel) where the suggested code/solution is:
```cpp
QGamepadManager* gamepad_manager = QGamepadManager::instance();
QList<int> gamepads;
int i = 0;
while (i < 10)
{
    a.processEvents();
    qInfo() << "get connected gamepads iteration : " << i;
    gamepads = gamepad_manager->connectedGamepads();
    if(!gamepads.isEmpty())
    {
        i = 10;
    }
    i++;
}
qInfo() << "connected gamepads : " << gamepads.size();
```
Where when the code initializes does a run through to count the number of gamepad devices.

I think that might be the solution however I think my problem now is I'm not sure how/where to inject that snippet of code.