vedderb/vesc_tool #126: compiling this for windows ?

roeebloch, 2021-04-17:
Hi,
How can I compile this for windows 10 ? - which compiler to use ?
Thanks,
Roee

Peemouse, 2021-04-17:
Hi, you can find it on @rpasichnyk fork

roeebloch, 2021-04-17:
no - there is no file there that run under windows 10 ?

Teslafly, 2021-04-17:
Look at the build_win_original_only script. This only builds in a Linux
environment, but should work in Ubuntu 16, which you can get working by
using wsl (windows subsystem for Linux). It's not super straightforward on
a windows machine. I am working on a way to make that better.

If you want precompiled binaries you can get them here:
https://vesc-project.com/vesc_tool

cammas, 2021-06-14:
Any chance dev releases could be kept here in Github?

Jfriesen222, 2021-07-02:
Anyone looking to do dev on windows can simply download Qt open source and use the minggw or msvc compilers. It builds nicely from Qt Creator.

Teslafly, 2022-12-06:
This probably fits into this issue:
https://github.com/vedderb/vesc_tool/issues/92
development documentation could be better.

Can this be closed?