vedderb/vesc_tool #129: Android: Enable RT Data Logging: Cannot Chose Log Directory

one1zero1one, 2021-04-24:
Hello, for android devices _without_  the possibility of using a sdcard (e.g oneplus 6), I did not manage to set a destination. 

Pressing "Choose log directory" is always empty and does not seem to allow browsing any directory. 

I've created a `/sdcard` folder, then I tried specifying `file:///sdcard`, but it errors with `directory not existing`. The only thing that does not error is `./`, but I didn't manage to find any files on the device.

Any idea what else should I try is super welcome, 

Thanks!

vedderb, 2021-04-24:
Is this with the version from google play? There is a known limitation in the current version that does not allow logging in the latest android versions. That is fixed in the development build and should work in the next update. Here are some details:

https://github.com/vedderb/vesc_tool/issues/116

Until the next version is released you can try the dev build apk from the forum thread linked in the issue above.

one1zero1one, 2021-04-25:
@vedderb thanks! I was indeed using the google play. The test build apk works fine, thank you!

closed 2021-04-25 by one1zero1one