vedderb/vesc_tool #136: PC VESC Tool stuck in dark mode

surfdado, 2021-05-13:
How do I change it to normal mode?

vedderb, 2021-05-14:
There is only a dark mode from now on to make it more consistent with the mobile version as parts of it are being integrated in the desktop version. @Jfriesen222 put in a few days of effort to make it, and I really like the way it turned out so far:
https://github.com/vedderb/vesc_tool/pull/133

At the moment there is no plan to support both a light and dark mode as it is easy to miss something and end up with unreadable text.

If you see any place with unreadable text or bad contrast please let me know and I will get it fixed.

vedderb, 2021-07-02:
Fixes with this:
https://github.com/vedderb/vesc_tool/pull/148

closed 2021-07-02 by vedderb

kubark42, 2021-09-29:
@vedderb thanks so much for the fix! See https://github.com/vedderb/vesc_tool/issues/165 for an explanation of why this was so important to me.