vedderb/vesc_tool #137: Motor winding resistance

Regelink, 2021-05-17:
I'm a bit confused how the 'motor winding resistance' as determined by VESC relates to what motor suppliers typically specify as 'Rm' or 'motor resistance'. If understood correctly, VESC determines the (average) resistance of one motor phase. Assuming a BLDC motor has its phases connected in delta configuration, the equivalent resistance of all phases in any (valid) commutation state, i.e. the line-to-line resistance would be 2/3 * winding resistance. [Line 1721 in mcpwm_foc.c](https://github.com/vedderb/bldc/blob/master/mcpwm_foc.c#L1721) also suggests that.

Now in VESC Tool, when I query the help for the **Motor Resistance (R)** parameter, a help message is displayed stating:

_"The motor winding resistance. Should be half of what is measured between two motor wires."_.

This message suggests that a star configuration of the motor phases is assumed.

I might be missing something, but to me it looks like the help message in VESC Tool should rather read something like: "The motor winding resistance. Should be half (for a star configuration of the motor phases) or 2/3 (for a delta motor phase configuration) of what is measured between two motor wires."

Perhaps it is also instructive to elaborate in the help message a little bit on what motor manufacturers typically specify in their datasheets as 'Rm' or 'motor resistance' and how this should relate to what is determined by VESC.

Regelink, 2021-05-18:
Wait a minute, the factor should be 3/2 (not 2/3). But then ... what is the factor 2/3 for in [line 1721 in mcpwm_foc.c](https://github.com/vedderb/bldc/blob/master/mcpwm_foc.c#L1721)? In any case, some more description in the help for this parameter would make it clearer.

vedderb, 2021-05-18:
Internally most motors have many poles that are connected together and affect the resistance, as well as a delta or start connection when combining the poles. All of that is abstracted away, and we only care about what it looks like when looking at the cables and pretending that the motor is a 2-pole star-connected motor with 120 degree phase offset. That approach work well for driving the motor properly.

The 2/3 factor in the code comes from that fact that the motor is transformed to a 2-phase motor using the Clarke transform in the code to make the math easier and less computationally intensive.

closed 2021-05-18 by vedderb