vedderb/vesc_tool #141: Boost charging through windings

cammas, 2021-06-04:
Hello [Vadym](https://vesc-project.com/node/1477) in two videos details how he charges from [Solar](https://youtu.be/jtmIfhdXWeU) and [PowerSupply](https://youtu.be/F64T9DeuCGA) through the windings of the motor to boost to end charge voltage. Has anyone worked on this? Does it exist? Is it possible to implement? Thanks !

vedderb, 2021-06-04:
That is easy. You can use DC motor mode, connect a power source with lower voltage than the battery voltage between one of the outer motor wires and the vesc and command a braking current (Polarity does not matter). That will use the motor inductance and power stage as a boost converter.

The disadvantage of using DC mode is that the commanded current will not be correct (the current will be 30% higher) as the current in the middle phase is not considered then. If you also disconnect the middle phase the commanded braking current will be the current drawn from the power supply. I have used this many times when I was too lazy to go to the garage and get my high power charger.

I'm thinking of making a dedicated charging app that also communicates with the VESC BMS for balancing with proper support in vesc tool, but first I want to finish the next firmware release.

jacopofranza, 2021-09-05:
Awesome work Benjamin!! This charging app would be really appreciated by many of us.

cammas, 2021-09-05:
It will be interesting to see more and more battery communication with the vesc. Maybe in the future it will be possible to get rid of charge circuitry on the bms and let the vesc take care of that. Essentially reducing BMS to mostly cell voltage and temperature monitoring.

Webst, 2022-02-22:
What's the status of this fine functionality?

cammas, 2022-05-22:
tried this today, DC mode, charger parralel to phase 1 and 3, brake current. nothing, current in stayed 0. motor current stayed same weird

A-damW, 2022-09-14:
@cammas You may have the incorrect 2 phases hooked up, look at the documentation for your (v)esc, find out which 2 phases on the (v)esc need hooked up for DC operation, I am guessing those are the correct ones.

livello, 2022-12-14:
@vedderb I tried but no luck!

https://user-images.githubusercontent.com/15374538/207696080-56fdc479-c8a2-465d-8807-4c005ce838ce.mp4

A-damW, 2022-12-14:
@livello
This may be of interest:
https://endless-sphere.com/forums/viewtopic.php?f=2&t=118585

kubark42, 2023-01-30:
@

> @livello This may be of interest: https://endless-sphere.com/forums/viewtopic.php?f=2&t=118585

Good on making it work. Your schematic makes it pretty clear:

```
MOTOR|-------------------------------PhaseC-----------------------------|VESC
MOTOR|-------------------------------PhaseB-----------------------------|VESC
MOTOR|---------------------|--cut---PhaseA--cut---|-------------------|VESC
                           |!!!!!!!!!!!!!!!!!!!!!!|
                           |-------DC Supply------|
```

Although it was still a little confusing for me until I realized what you were showing was the DC power supply's positive and ground placed in series. Maybe my ASCII art below helps people see it from a second vantage point:

```
MOTOR(PhaseC)|------------------------------------------------------------|VESC
MOTOR(PhaseB)|------------------------------------------------------------|VESC
MOTOR(PhaseA)|----------------|NEG-------DC Supply------POS|--------------|VESC
```

livello, 2023-05-24:
> > @livello This may be of interest: https://endless-sphere.com/forums/viewtopic.php?f=2&t=118585
Thank you! 
I'am using this way with my power supply 43 volts to charge 13s lifepo4 battery. 
But this is dangerous! Power supply may affect full battery voltage!!!!
I burn't two of them. 5 volts and 12volts power supplies!!!