vedderb/vesc_tool #143: Android: Realtime Data Logging doesn't work

zippymcbonk, 2021-06-19:
Tapping "CHOOSE LOG DIRECTORY..." shows white blank space where the file list should be, meaning it's probably silently ignoring an error. Then tapping ".." in top right corner goes to the root of the filesystem. Attempting to navigate to a mounted sdcard shows blank white space again.

Attempting to input a file path manually, e.g. "file:///storage/emulated/0/vesc" gives "Output directory does not exist". When I go and create the directory, the error changes to "Could not open file for writing.". That would hint at a permission issue.

The App has the storage permission and I've just tried to disable/enable it a few times, didn't help.

FWIW I have found the app's home directory at "/data/data/vedder/vesctool/files/", this is the directory it logs to when you've got a path like "./log" set in the app (which was the default for me). So I can log and find the files, but it is hugely inconvenient to access them there (rooted system is likely necessary).

A forum thread with more people reporting the issue: https://vesc-project.com/node/2772

vedderb, 2021-08-20:
Is this one the same issue?
https://github.com/vedderb/vesc_tool/issues/116
If it is it will be fixed in the next release. Until then you can give the apk from the beta forum thread a try,

riddimrider1, 2022-04-18:
Not sure if it was the same, but it is now fixed (sorry about the delay in responding). Also the UI is much nicer now, thanks!

closed 2022-04-18 by riddimrider1