vedderb/vesc_tool #157: VESC Tool closes with version check failure when vesc-project.com is not online

TechAUmNu, 2021-08-19:
https://github.com/vedderb/vesc_tool/blob/b0632c42e896175f9d731be5f146f0202a0fe17c/utility.cpp#L129

No handling for vesc-project.com being in some broken state, so vesc tool just closes itself after 10s.

Workaround is to disconnect network from the device running vesc tool.

kubark42, 2022-01-30:
Looks like this is a blocking call: https://github.com/vedderb/vesc_tool/blob/b0632c42e896175f9d731be5f146f0202a0fe17c/utility.cpp#L138

So if the HTML response is dragged out, then this will block. 

Also, there's not sanity checking of the returned result. A garbled response might cause unexpected behavior.

Teslafly, 2022-12-06:
Well, vesc-project.com is down again and the current dev version build with qtcreator appears to still work. So this is probably fixed?