vedderb/vesc_tool #159: Status: No firmware read response with Bluetooth - I can't USB because I glued my ESC

fredvm, 2021-09-01:
Hi 

I glued my ESC for marine application with a BLE connected to the board. I didn't keep a cable for a USB connection.
I made maybe a mistake in the configuration with the android App, and now I have This message : Status: No firmware read response /  Could not read firmware version. Make sure that the selected port really belongs to the VESC. 

Is there any possibility to force the connection and to update my VESC via bluetooth... Or there is no way, and I must put my glued ESC in a trash ? ;) 

Thanks 

Fred

[image]

surfdado, 2021-09-02:
If you accidentally changed the app to "None" or any other app that doesn't include UART functionality then you're out of luck. There's no way to force a connection because the BLE module still wouldn't be able to get any communication to the VESC because the UART pins are inactive. USB is the only way to recover from that I'm afraid.

Teslafly, 2021-09-02:
If you have access to the swd programming port, flashing the firmware will return settings to default. You may be able to dig in just enough to get to it. You only need the dio and clk pins.

Another alternative would be if you have access to the can port. You could use another vesc to reset it to defaults.

fredvm, 2021-09-02:
Thanks guys for the answers - I will try to dig at the good place ;) - I close the issue, I have enough information

closed 2021-09-02 by fredvm

fredvm, 2021-09-04:
I find the can port. But I can't detect it with an other vesc (all the connection are good with the same power+ and ground) . Teslafly are you sure that I can detect my vesc by CAN bus with my problem? 

⁣Télécharger BlueMail pour Android ​

Teslafly, 2021-09-05:
I just tried it, Using a vesc 6 to update a hw 4.2, and it worked fine.

My steps were:
1. Connect the 2 vescs can ports together (gnd, canH/canL)
2. Power both vescs with battery power. Connect usb/vesc tool desktop to the "unpotted" vesc. (vesc 60_mk4 in my case)
3. Click "scan CAN". Your second vesc should be detected in the "CAN-Devices" box (see screenclip)
4. Double click on vesc we want to upload firmware to. There should be a darker grey box showing it's selected. (vesc 410 for me) 
5. Go to firmware page and upload firmware. (Actually at this point, you may just be able to enable the uart app in settings again)

[image]

I hope you can get this to work.

If you do end up potting any more vescs, I recommend breaking out the usb, swd pins (clk, io, rst), and maybe CAN if possible. They make capped waterproof connector pigtails you should be able too use. an 8 pin should be enough (or more smaller ones)
If you have the swd pins you can recover from anything other than physical damage.

fredvm, 2021-09-06:
Finally it was not the good spot... I did it with the CAN

Thanks guys 😁

⁣Télécharger BlueMail pour Android ​

Teslafly, 2021-09-06:
Glad you were successful!