vedderb/vesc_tool #186: [Suggestion] Disable the arrow key control whenever the user clicks in a text field

kubark42, 2021-11-17:
Several of us have noticed our propensity to enable arrow-key controls, and then absentmindedly use the arrow keys while trying to edit a text field. The result is a motor which leaps to life unexpectedly.

The best accidental outcome is that nothing breaks, nothing explodes, and no one gets hurt. However, the worst-case scenario is pretty grim. 

It seems that there are few, if any, situations where the user would like the arrow key to continue functioning while the user is actively editing code. 

My suggestion is that the UI deactivate the control mode immediately upon receiving this conflicting usage of the arrow-key. 

In the event the system autonomously deactivates arrow-key control, the user should be notified. One way to do this is by flashing the arrow-key button.

Jfriesen222, 2021-12-02:
This is a very good idea I will investigate

Jfriesen222, 2021-12-02:
https://github.com/Jfriesen222/vesc_tool/commit/3280f6251a5178c4f9d71d8345112f74172e993d