vedderb/vesc_tool #197: Windows version Bluetooth connections not possible

surfdado, 2021-12-02:
Bluetooth connections from a Windows PC don't work. Not for me, not for anybody I know of. From Linux however it is no problem... So Windows users are forced to use the TCP bridge via Android app. Which works just fine, if you own an Android phone...

Maybe this is a known issue and maybe it's by design, but I thought I'd file an "Issue" anyway :)

Teslafly, 2021-12-02:
This is due to qt not supporting bluetooth on windows on earlier versions. With qt5.15 it does have native support, but i needs to be enabled and I think that bluetooth works different on Linux vs windows in the backend so you have to handle both cases.

https://doc.qt.io/qt-5/qtbluetooth-index.html

So it is probably possible to enable Bluetooth support in windows now, but it will take a bit of work. I have looked into it but do not understand qt well enough to get that working at this point. 

Though an error message saying that Bluetooth doesn't work on windows would definately be nice until support is added.

Teslafly, 2021-12-02:
looks like some more info here: https://www.vesc-project.com/node/2606

surfdado, 2021-12-03:
I agree that a little popup message would be a huge improvement and save many hours of trial and error, frustration, and self doubt :)

Jfriesen222, 2021-12-03:
if you compile the tool with MSVC 2017 the bluetooth works. It's just not possible (or at least easy) to cross compile from linux though which makes more work for Ben. For this release I might make a MSVC binary for Ben to use. I wonder if windows 11 will be less annoying with BLE.

Teslafly, 2021-12-03:
Would it be possible to set up a ci pipeline that spins up a windows vm and spits out an exe whenever a merge to dev or master is made? That would probably make it a lot easier. And also easier to test new developments if you don't have qt installed.

Jfriesen222, 2021-12-03:
that's above my paygrade

Teslafly, 2022-11-06:
Bluetooth has been enabled for windows and works with windows compiled qt in the current beta. The beta version from vesc-project.com works too. and I assume that's the cross compiled version. 

Note, you must connect to the vesc bluetooth module in windows bluetooth manager first. Then scan/connect in vesc-tool.

Current issues with windows bluetooth:
- You can only connect once, then you have to restart vesc tool to connect again. Other qt developers have noted this so it may be a core qt issue or a bug that's easy to have.
- Windows doesn't seem to provide the service uuids so it's impossible to filter non vesc ble devices without connecting to them.
- Multiple vesc bluetooth adapters don't show up in windows bluetooth manager. Windows can only see 1 or none at all when there is multiple.

surfdado, 2022-11-07:
Wow, that's a lot of issues hahaha - but nonetheless this is amazing news, thanks guys!!!