vedderb/vesc_tool #203: ADC status missing in RT Data

LAP87, 2022-01-15:
As the title specify, ADC 1 and 2 status is gone from the RT Data fields beneath the gauges in the test 30 version of Vesc Tool on android.
Would love to have it back, since it's quite important to us who ride balance vehicles :)

Best regards,

vedderb, 2022-01-15:
Are you referring to the adc inputs on the comm port? I don't think they have been there before either.

surfdado, 2022-01-15:
There used to be 4 screens - it looks like the entire RtDataBalance screen has disappeared. That's the one that shows ADC1/2 for the foot sensors on our balance vehicles. It looks like @Jfriesen222 made a change recently that's supposed to only show this screen if the balance app is enabled, but now the screen doesn't ever show up

vedderb, 2022-01-15:
Ah, the balance page. I tried it just now with the latest version, and it works for me. Are you using the latest build from here?
https://vesc-project.com/node/2859

surfdado, 2022-01-15:
You're right - the issue got fixed in the latest version. I can see the balance screen again.

closed 2022-01-15 by vedderb