vedderb/vesc_tool #208: SWD programmer VESC SIX MkV missing

Talabo, 2022-02-08:
Hi!

While fixing a bricked VESC 6 MkV with another VESC (connecting with the SWD ports) I saw that in the SWD programmer under "included firmwares" the VESC 6 MkV (Mk5) was missing, and I only saw Mk4 and Mk3.

I fixed the VESC 6 MkV with Mk4 which makes the VESC Tool think it's a Mk4 after fixing the bricket VESC. I did a work-around by uploading a custom FW version which I have decleared for Mk5, so the VESC Tool knows it's a Mk5 and allowing me again to upload the original Mk5 FW version again through the normal procedure in VESC Tool.

For convenience the Mk5 should also be included in the "Included Firmwares"-list under SWD programmer.

Teslafly, 2022-11-06:
is this fixed now?

vedderb, 2022-11-07:
Should be fixed here:
https://github.com/vedderb/vesc_tool/commit/4ef8598dca3cfb014520ebb64efcbdde7f61508e

closed 2022-11-07 by vedderb