vedderb/vesc_tool #210: Firmware update custom file on Android

danilolattaro, 2022-02-21:
The latest 3.01 stable version of Android app doesn't show any files when browsing for for a custom firmware.
The "choose file" button opens the storage navigation, the folders are shown and can be selected, but when accessing the correct path where VESC firmwares are stored no files are shown and therefore a selection can't be made.
The 3.01 beta with previous aesthetic was working fine, but the newer good looking app seems to have this feature broken.
The app was given full permission to storage and files.
Running on Android 11.
[image]
Screenshot without connected VESC only for illustration purposes. As can be seen, a path is shown on the top but absolutely no selectable file.

vedderb, 2022-02-21:
The problem is that even with file permissions VESC Tool can only access very few locations since the latest google play store requirements. Getting around that is almost impossible.

What worked for me for making the file visible to VESC Tool is putting it in
Android/data/vedder.vesctool/files

danilolattaro, 2022-02-21:
Had to download another file manager (Total Commander) to manage and copy the bin to that path, but can confirm: also worked here.

danilolattaro, 2022-02-21:
How to deal with this now?
Workaround is plausible but may be difficult for other users.
I guess have to leave the issue open until the Google play gives a way

Teslafly, 2022-05-03:
I wonder it you could make it so you can "share" a file with vesc tool and it copies it there?