vedderb/vesc_tool #239: Cannot connect to VESC tool mobile with bluetooth

pistolriver, 2022-07-23:
Thanks for your help! I tripped over the usb cable connecting my pc to the my Maker XH200 VESC and broke off the connector-ARGH! So I thought I could use Bluetooth.
I cannot get my Maker X or Flipsky Bluetooth modules to connect to both the pc version or the mobile app. I keep getting prompted for the UUID. I  tried switching the RX and Tx wires and got a firmware error. I looked in my Bluetooth settings and tried the driver GUID it it didn’t work.
[image]
[image]

Teslafly, 2022-11-06:
is this still an issue with the current version of the vesc tool app? I have no issue connecting at all.
are you using the vesc tool apk downloaded from vesc-project.com?

gregers666, 2023-04-25:
In my setup there is possibility to connect via Bluetooth with full VESC working on Linux laptop, but no possibility to connect with mobile app.
Mobile app doesn't show any device during scanning.
I use mobile app downloaded from https://vesc-project.com/vesc_tool 
It is possible to pair and connect using Bluetooth Serial Terminal from the same mobile.
Setup consists of 75100 Alu Board  connected JDY31-SPP (tested UART1 and UART2 both work the same)
On Linux full application connection consists of following steps:
- pairing with JDY31
- connecting JDY31 to rfcomm0
- selecting rfcomm0 with 115200baud from VESC
- connection is estabilished and everything works perfectly.

BTW... I needed BT connection because connecting Arduino via CANBUS disconnects USB connection with laptop, even if working from totally separated power sources (batteries).

Teslafly, 2023-04-25:
Vesc bluetooth only works with a custom bluetooth adapter running custom firmware. 

Either the vesc express over can
https://trampaboards.com/vesc-express-with-canbus-power-cable-2-p-35329.html

Or an nrf board over uart. 
https://trampaboards.com/vesc-connect-nrf-wireless-dongle-with-integrated-antenna-p-29139.html

https://www.amazon.com/Flipsky-Wireless-Bluetooth-Skateboard-nrf51_vesc/dp/B0894M3Q4X