vedderb/vesc_tool #241: Where can I find PID position control from source code?

QZJGeorge, 2022-08-10:
Hi,

I am trying add a current limit on the position PID controller to prevent my motor from jump start. Could someone show where this part of code is located? Thanks.

Aquaharmonics, 2022-09-06:
Hi,

Not sure on the exact use case but have you tried changing the positive/negative ramp times?

Teslafly, 2022-09-06:
What control input are you using? Ppm has safe start modes that should wait for the stick to be close to the motor position.

Teslafly, 2022-09-06:
Logic for angle check before enabling motor found here: 

https://github.com/vedderb/bldc/blob/master/applications/app_ppm.c#L352

Teslafly, 2022-11-06:
can this be closed?

QZJGeorge, 2022-11-06:
Yes thanks please close it.

Teslafly, 2022-11-07:
I can not close this. Only you, the issue creator, or the the project author/dev can close issues. You should have a button at the bottom of the page to close the issue.

closed 2022-11-07 by QZJGeorge