vedderb/vesc_tool #243: TS5700N8501 implementation

Aquaharmonics, 2022-09-05:
Hello,

I have been attempting to implement a Tamagawa TS5700N8501 encoder[(chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/http://www.bianmaqi.com/uploads/201803/2018032819044985397.PDF)](url)for motor feedback. 

I have not seen any other implementation of this except an attempt on the forums here: [https://vesc-project.com/node/2661](url)

Basically just wanted to see if there are some obvious things I am missing in hardware or software setup, and if anyone else has been successful using this particular encoder. I am getting a continuous fault: ENCODER_SPI.

VESC TOOL Configuration:

Under MOTOR SETTINGS>GENERAL >GENERAL
I have SENSOR PORT MODE set to " TS5700N8501 (multiturn)"
I have "ABI ENCODER COUNTS" set to "131072" . I do not believe this is actually used in configuring this particular encoder, however, based on reading through "enc_ts5700n8501.c" [https://github.com/vedderb/bldc/blob/master/encoder/enc_ts5700n8501.c](url), the number of encoder counts per turn appear to be hardcoded.

[image]

Under  MOTOR SETTINGS>FOC>GENERAL
I have  SENSOR MODE set to "ENCODER"
[image]

HARDWARE Configuration:

I have wired the encoder to the "COMM" port on a VESC 75/300 V3 : The blue wire to "TX", the blue/black wire to "RX", Red to "5V" , and black to "GND". 
[image]

What I am seeing:

Using the VESC terminal command "encoder" I typically get something like this as an output, with the encoder plugged in:

SPI encoder value: 16364479, errors: 5325513, error rate: 99.940 %
TS5700N8501 ABM: -1571, SF: 11001011, ALMC: 11111111

If I move the encoder shaft a small amount, say ~5 to 10 degrees, the values do change, but do not seem to update correctly:

SPI encoder value: 10411455, errors: 5782452, error rate: 99.928 %
TS5700N8501 ABM: -1059, SF: 11001011, ALMC: 11111111

If the encoder is unplugged the last values recorded stay, and the errors keep counting. 

I have tried two different encoders wired the same way, and also with the blue/black and blue wires swapped on the "TX" and "RX" pins. With the pins swapped, no change in encoder position is seen.

Below is a screenshot of the RT data screen, showing that every now and then it seems like the output of the encoder is wrapping around and you will see encoder position jump.

[image]

I did see per the forum link that a line drive may be needed, but there is another encoder in the VESC tool MT6816(SPI) that also uses a Serial Port Interface, which looks like what the TS5700N8501 uses per the following document:[https://ams.com/documents/20143/36005/AS5047P_DS000324_3-00.pdf](url)

Thanks for any additional help on this, I will keep reading through the code hardware documentation and see if there is something else I am missing.

-Alex

Teslafly, 2022-12-06:
See the `TS5700N8501 Encoder` section of the sensor port mode help page in vesc tool.

You need an rs485 tranciever with the rx/tx enable/direction pin being driven by the adc1 pin.

[image]

Aquaharmonics, 2022-12-06:
Hi @Teslafly,

Thanks for pointing this out, I completely missed this in the help section of VESC Tool. This makes a lot more sense now.  In the setup I am using, I was planning to use UART with a Discovery board, and in testing, UART with ADC1 so I will have to think about that one a little bit. It seems like CAN is still available, however. Perhaps initially even using two VESC's, one for the ADC1 input and communicate over CAN to the second VESC that is driving the motor and using the TS5700N8501 for commutation. 

A follow up question then:

It seems like for the encoder options such as the TS5700N8501, or Sin/Cos, both higher resolution absolute encoders and desirable for more accurate commutation at lower speeds, the ADC 1 has to be given up. Is there anything special at the hardware level about ADC 1 in particular? Could ADC 2 be used instead for the input of the Sin/Cos encoder? Or, could ADC 2 be used for the throttle/current control input?

I believe this can be closed. Thanks again!

Teslafly, 2022-12-06:
Adc2 could be used instead I think.(as long as it is bit banged and not using some periphrial) You would have to compile your own firmware after changing the pin though. An swd or hall port pin may be a better choice though as those are less used than the adc pins. 

If you don't need the resolution/accuracy, the spi hall encoders like the ams5047 and tle5012 can use the hall sensor port and leave the adc/uart pins free for other uses and work just as well for motor commutation.

Only the issue creator or a repository maintainer can close an issue. I am neither.

closed 2022-12-06 by Aquaharmonics