vedderb/vesc_tool #248: Bluetooth connection issue Mac desktop app

marcofloato, 2022-09-30:
Hello,

I’ve discovered that before you can install the Mac desktop app and connect to your VESC via Bluetooth, that you first have to install the iPad app to your Mac and first connect via Bluetooth there. Once you’ve done that, you can install the desktop app and connect to Bluetooth. But, once you close the desktop app, you cannot reconnect via Bluetooth until you’ve opened the iPad app and connected to Bluetooth there, again. If you don’t, when you try to open the desktop app and connect to Bluetooth, the app will shut down. It will repeat this every time, unless you repeat the steps above. It’s very strange, but I can repeat it every time.

lidocaineus, 2022-10-30:
Can confirm this exact behavior. Attached is the error log generated by the crash when you don't do the iPad app thing.

[crash_log.txt](https://github.com/vedderb/vesc_tool/files/9895080/crash_log.txt)

Teslafly, 2022-11-06:
this possibly seems like an issue with the ble dongle firmware. that or the mac needs to see some ble traffic first?
but it shouldn't crash.

is this still an issue with the current beta if that's possible for you to test?

it would also be interesting if you have the same issue using this and an esp32 for the ble dongle.
https://github.com/thankthemaker/rESCue

lidocaineus, 2022-11-08:
Is there a macos binary of the beta build available? Doesn't look like it's in the zip archive with the other betas. It does seem like some sort of bluetooth issue as the behavior appears even worse on Ventura - just scanning for bluetooth signals causes lockups and crashes. TCP bridge connections work fine.

lachlanhurst, 2022-12-01:
I just confirmed this. The latest build available from https://vesc-project.com/ crashes instantly when attempting to connect via Bluetooth.

I then cloned & built VESC Tool and Bluetooth seems to work fine.

marcofloato, 2022-12-01:
> I just confirmed this. The latest build available from https://vesc-project.com/ crashes instantly when attempting to connect via Bluetooth.
> 
> I then cloned & built VESC Tool and Bluetooth seems to work fine

lachlanhurst, 2022-12-01:
@marcofloato "cloned" is really just downloading the source code. Most developers would use git to do this, but you can always [download it from here](https://github.com/vedderb/vesc_tool/archive/refs/heads/master.zip).

"build" just means compile the application, this is a bit more complicated. I used [Qt Creator](https://www.qt.io/product/development-tools), and found it straightforward to open up the vesc_tool.pro project file, then build the application. You may also need Xcode installed too, I'm not 100% on this.

Good news is that it does seem this issue will be fixed in the next MacOS release.

lachlanhurst, 2022-12-03:
So there does seem to be a bit of an issue related to permissions here.

When you run VESC Tool for the first time you get the following popup.
[image]

Click ok and the Bluetooth connection will work fine for that session of the VESC Tool. However, if you shut down the VESC Tool and start it again it will crash when you attempt to connect to the VESC.

There is a workaround though. You can open up the system preferences and remove the VESC Tool entry from the list of applications allowed to use Bluetooth. Next time you start the VESC Tool you'll get the permissions popup shown above.

[image]

This workaround works for v6 and the old version of the VESC Tool

lidocaineus, 2022-12-03:
Can confirm that workaround does NOT work for me, even after deleting the iOS build of VESC Tool, which appeared to be causing name collision issues when installed at the same time.