vedderb/vesc_tool #252: Keyboard control command

Kupofty, 2022-10-18:
Hello,

I'm having issues controlling my motor with the arrows on the keyboard after activating the "keyboard" option.
Sometimes I can go to 100% PWM (the PWM increases the longer I push the right/left arrow) but sometimes it caps at ~ 20%.
I'm using "vesc_tools 3.0".

Is there any known solution to that problem ?
Thanks.

joonashopman, 2022-10-30:
What is your maximum motor phase/battery current? This could be the limiting factor in the speed your motor achieves.

Kupofty, 2022-10-30:
The motors datasheet says 20A max and I use a voltage generator. It can deliver much more current than the motor requires. Vesc tools parameters are set according to the datasheet. The motor has no issue going 100% PWM by sending 100% duty cycle command (button on the down left corner). 

The problem really is with the keyboard arrows. Sometimes it works fine and I can push the motor to 100% and sometimes it blocks at 12% or 20%. If I restart the pc or vesc tools several times it eventually works again.

gregoplex, 2022-11-12:
Ditto! 

Did your motor spin fine during the motor setup/wizzard but only worked "sometimes" with the vesc_tool connected and attempting to test with the keyboard input!? 

I noticed it seemed to work best when nobody else is watching and "never" works during demos (I have personally found). I updated my vesc_tool and re-flashed firmware several times as well as trying several different batteries and power supplies that were more than adequate for the task. I am currently on 3.01, but was likely on 3.0 before updating. I know the keyboard input to test run the motor used to work just fine on much older releases...years ago.

After checking, rechecking, and triple checking every setting I could think of, I finally moved on to setting up my ppm control and the motor spins on command exactly as it should. current is within spec, and everything else works as it should, so I have since moved on. Sorry that I don't have any specific lines of code to call out, but this might deserve some further investigation by those more knowledgeable.

vedderb, 2022-11-12:
The keyboard control is supposed to behave a bit "strange" in duty cycle mode (left and right arrow keys). If you press right it will ramp up to 30% or so and stay there. Then, if you continue holding right and press left at the same time it will continue ramping up to about 90% as long as you hold both keys. This is a very comfortable mode for me to test the motor when I have it on the bench.

closed 2022-11-29 by Kupofty