vedderb/vesc_tool #256: Allow profiles or similar to save entire motor/app configs on mobile

Teslafly, 2022-11-06:
Since it is not possible to save/restore configs on the mobile app. (or at least not clear?) it would be nice if there was a profiles like feature to save the state of entire configs. 
being able to tinker around with settings and then reset them to a known good configuration is really useful on the fly.

Especially the balance app where there's a lot to tweak and go wrong. or settings you would want different beyond current for different types of riding.

I tend to want to play with motor configurations a lot and need to revert settings reasonably often.

even if saving configs as files was possible, easy access with a big button like profiles would be very nice.

Teslafly, 2022-11-06:
This video mentions an ask for a feature like this.
https://www.youtube.com/watch?v=AWp4XteFnXo

vedderb, 2022-11-07:
The latest betas can export and import xml-files using the android file browser. It is also possible to use the backup and restore buttons, but that is based on uuid and will always replace the old config for that uuid. There are also the profiles, which are intended to include settings that you would want to change quickly while you are outside and not exceed the limits of the original config (which is why they are percentages).

I could make a config manager that stores everything using a name (instead of uuid) locally that lets you reload the configs for that name. Not using the file browser and just leaving that in the cache for the app makes it much easier as android won't fight you in the same way then. Then if you want to export those configs you have to use the xml save and load dialogs.

The problem with sharing configs is that many things, especially sensor and encoder settings, won't work the same way on other setups. Therefore when it comes to the balance app it would be good if you agree on a subset of parameters to share that are relevant for the balancing. I sent  @Mitchlol a video the other day where I showed how to pick any set parameters from any config and save them in a file and how to reload that file. As that approach uses the config-code it is robust against version changes and having different parameter sets in those files.

For the balance app I think it would be useful if the qml-page also has a tab with some quick-settings that people are likely to tweak while riding. I'm not sure what those settings would be or if that even is a good idea (probably @Mitchlol knows), but it is quite easy to make a page like that.

Mitchlol, 2022-11-07:
Yes, I intend to implement a balance app shared subset exporty thing. There is just a lot of things to do, and I'm not as fast as Ben :smile: