vedderb/vesc_tool #277: SIN/COS encoder parameters change not taken into account until reboot

totosolat, 2022-12-06:
Hi,

I noticed that I need to reboot the board before the SIN/COS encoder parameters change take effect. (amplitudes, offsets)

Is there a reason for that? If not, I think it should be fixed. I could help on that if we agree on the expected behavior.

Teslafly, 2022-12-07:
The expected behavior is for all parameters to take effect when mcconf parameters are written to the vesc. So yeah, this is probably a bug if reproducible.

vedderb, 2022-12-07:
Should be fixed here
https://github.com/vedderb/bldc/commit/eade5ce18b60c1e51708466eadae9e705401ecb5

totosolat, 2022-12-08:
Thanks a lot, I will test it soon and let you know!

totosolat, 2022-12-09:
Hi,

I just tested, I can confirm the parameters are updating fine now.
Thanks!

closed 2022-12-09 by totosolat