vedderb/vesc_tool #284: Vesc tool download from archive crash on windows 11

cedrox, 2022-12-12:
When download old firmware from archive, soft crash (at the end of the download)
> download / Re-download Archive
[image]

vedderb, 2022-12-12:
That is probably because the archive is quite large now and the download stores it in RAM at least twice before writing it to disk. I just made an update so that it writes the file to disk as it downloads it instead:
https://github.com/vedderb/vesc_tool/commit/60f090c4794728e2b1e4dbaba7519e451b84b4cd
The beta is updated if you want to try it out.

I'm planning to make a 6.02-release quite soon with mostly bugfixes and small improvements like this one. There are so many new features in 6.00, so I think a few more things will show up now that more people are using it.

cedrox, 2022-12-12:
I will wait the release to install it, test it and close the issue as I don't have my dev env setup to compile the last release...
Thanks Mister Vedder !

vedderb, 2022-12-12:
The beta-build can also be downloaded from the website. Under purchased files there should be a link to the latest beta-build with files for Linux, Windows and Android:
[image]

cedrox, 2023-01-16:
Thanks for confirming. Close

closed 2023-01-16 by cedrox