vedderb/vesc_tool #285: res/firmwares/res_fw.qrc ??????

karu2003, 2022-12-13:
I've been trying to build a new release for 3 hours now and I'm getting a warning. That the file was not found.
res/firmwares/res_fw.qrc

jaykup26, 2022-12-13:
I think you need to compile the firmwares in the bldc repository, then run the package_firmware.py script to create the res_fw.qrc file.

karu2003, 2022-12-14:
Two weeks ago this was not necessary. It is very uncomfortable. transfer from PC to PC.

vedderb, 2022-12-14:
Yes you need to build the firmwares. This has been necessary for the past 8 months or so, otherwise you would get missing and out of date firmwares. Only difference now is that I removed those old firmware from the repository.

This is a necessary approach as the git history would be huge if I updated every single firmware for all commits. That is why the package_firmware script can generate the resource-file too - no need for any manual editing when adding new hardwares.

Another approach is to comment out res_fw.qrc from the project file and build vesc tool without the included firmwares.

closed 2022-12-14 by vedderb

FPSychotic, 2023-05-02:
> Hi where I can comment out that? there is not project file.

r00t-, 2024-05-02:
you can add
```
CONFIG += exclude_fw
```
i.e. in build_lin

as added in https://github.com/vedderb/vesc_tool/commit/5b1ad1ddb2848060f2c1b647910b115e5e6a2563