vedderb/vesc_tool #286: VESC tool won't run input wizard after downgrading firmware through archive feature

jaykup26, 2022-12-14:
If you use the new Firmware -> Archive feature to downgrade from 6.00 to 5.03 or older, the input wizard will not run when using the 6.00 tool.

[image]

If this could work on at least firmware versions 5.01, 5.02 and 5.03 using the 6.00 tool or later, I think that would cover most use cases.

vedderb, 2022-12-14:
I hope this fixes it:
https://github.com/vedderb/vesc_tool/commit/75c64a7e37221bd2b00323a358f7eed24a683318
The beta is building now, should be done any moment now if you want to try it out.

jaykup26, 2022-12-14:
Tested and working great with latest commit https://github.com/vedderb/vesc_tool/commit/3540ae81cbcea9b85106dbe11f01c7bb0942d88e

closed 2022-12-14 by jaykup26