vedderb/vesc_tool #289: lost /res/firmwares/ directory

wangkj123, 2023-01-01:
ubuntu 16.04
git clone https://github.com/vedderb/vesc_tool
cd vesc_tool
./build_win
Info: creating stash file /home/ubuntu/vesc/vesc_tool/.qmake.stash
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
qmlimportscanner: No such file or directory: "/home/ubuntu/vesc/vesc_tool/res/firmwares/res_fw.qrc"
/opt/mxe2/usr/i686-w64-mingw32.static/qt5/mkspecs/features/qt.prf:307: Error parsing JSON at 1:1: illegal value
Project ERROR: Failed to parse qmlimportscanner output.

MrSurly, 2023-02-21:
I'm seeing something very similar:

```vesc_tool$ ./build_lin
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
WARNING: Failure to find: res/firmwares/res_fw.qrc
/usr/lib/qt5/bin/rcc: File does not exist 'res/firmwares/res_fw.qrc'
/usr/lib/qt5/bin/qmake -o Makefile vesc_tool.pro -config release 'CONFIG += release_lin build_original'
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.```

BiuBiuHou, 2023-02-22:
You should add res_fw.qrc manually. Then add your firmware .bin file in the res_fw.qrc file. Run the project.

ghent360, 2023-03-26:
Is there an example, what am I supposed to put there? vesc firmware binary?

BiuBiuHou, 2023-03-29:
Right. Put vesc firmware binary