vedderb/vesc_tool #295: [Feature request] Use multiple position sensors (SIN/COS + AS5048A) simultenaously

totosolat, 2023-02-14:
Hi,

I use a VESC 6 edu for controlling robotic actuators and I would like to do the following:
-Use SIN/COS sensor on the rotor side for FOC control (which I have already done, it works well)
-Use an AMS AS5048A as absolute output position (after a reduction ratio or 100) in order to know the output absolute position I don't have with the SIN/COS on the rotor side. I only need to be able to access this position value from CAN and/or application, not for FOC control. It could also be used only one at sartup to get absolute output position only once and use SIN/COS sensor after getting absolute reference. 

Both configurations work fine but I can't make it work together at the same time, the configuration tool does not allow this.

Using only the output AS5048A for FOC and absolute output position is not a good option since the limited electric angle resolution and errors due to the reducer does not allow for good FOC control.

Thank you

Teslafly, 2023-02-14:
This seems like a rather unique use case and I would encourage you to try and implement this feature on your own if it is important to you. Otherwise if your gearbox doesn't have a lot of backlash you can use the multiturn angle division setting in the pid tuning as your gearbox ratio to map the motor multiple turns to your output angle.

Jerry3012, 2023-04-19:
I have exactly the same needs and may I ask if you have already started working on this feature?

totosolat, 2023-04-19:
Yes, I am working on it right now.
-We are using sin/cos encoder for FOC control
-We are developing custom user app which reads the AS50xx encoder. We use this encoder value to set the PID pos offset parameter.
-We also created a lisp function to access the encoder value.