vedderb/vesc_tool #307: Beta program for iPhone App

surfdado, 2023-03-16:
@Jfriesen222 

It would be really helpful if there was a beta program - currently most iPhone users are more or less disqualified from testing beta VESC firmware/tools. Having an iphone version being released on TestFlight, keeping more or less up to date with Benjamin's VESC releases would be perfect!

vedderb, 2023-03-17:
I think the answer is simply that if you want to participate in the beta-testing you either have to use something that is not apple or compile it yourself. Supporting a device that does not let you install software without the permission of the vendor is not something that I'm very enthusiastic about. It really shows that even though you paid for it they still own and control it, which is the complete opposite of the ethos behind the VESC Project.

surfdado, 2023-03-17:
I don't personally know all that much about Apple's ecosystem but I do know that beta versions can be pushed to up to 10k users without any of Apple's involvement. If Jeff doesn't want to do it I could give it a try. But that would be awkward because it would be under my Apple account. 
Also, is the source code that Jeff uses publicly available? I am able to build iPhone apps but the VESC Tool app isn't one of them. It looks like Jeff has at least some XCode project files  that aren't part of the vesc_tool code base. Not sure what else he had to do to make it work

Jfriesen222, 2023-03-17:
All the code used to build is in the repository. You have to install xcode and qt creator on a mac and then build for iOS, I think you also need an apple developer account to deploy to your own devices. At this stage I don't think it is worth the effort of running and maintaining a beta through test flight as most everything gets tested through other platforms already and I do the testing for iOS here myself. Having someone else manage a testflight release would just create more communication and work in my view, I think as it is now is fine for the near future.

surfdado, 2023-03-17:
I do have the Apple Developer account, but I'm not very experienced with XCode/Apple. I'm doing a lot of work with the VESC beta versions but am forced to label them "6.0" so I can keep using my iPhone. So I would LOVE to have a beta build of the iOS app. I'd very much appreciate if you could point me to some build instructions.

Note: I haven't installed Qt yet - is QtCreator what's used to build the whole iOS app? Because I don't see an XCode project file in the code base. Thanks!!

Jfriesen222, 2023-03-17:
Yes, you first build from qt creator and then it creates an xcode project during the build. Though generally you do not need to open xcode to deploy to the phone. Just download Qt creator installer from the website check the box to install Qt 5.15 for iOS when you install it then just open the vesc-tool project in qt creator select ios as your target and it should build. If you just do some general googling on how building Qt projects for iOS and deploying to iphone works it will probably be more effective than me explaining it second hand.

surfdado, 2023-03-17:
Sounds great, will give that a try, thank you!

closed 2023-04-06 by surfdado