vedderb/vesc_tool #311: Log to CSV - Default folder is invalid / Lack of user feedback

surfdado, 2023-04-06:
The folder provided as the default folder either doesn't exist or isn't writeable, so prior to being able to log one needs to manually select a valid folder, which is really tricky to do.

Furthermore, the CSV checkbox happily lets you check it and won't complain that the folder doesn't exist, so the crucial first logs of many users end up non-existent.

Btw, ever since I've started using the VESC App on Android this has been the same issue.

unity-punk, 2023-04-06:
Also having this issue on Android, hard crash when I try to enable logs.

Browsing to a new folder leads to a broken interface, forcing a reset of the app!

nitrousnrg, 2023-04-13:
Similar problem -probably worse- on iOS.

@Jfriesen222 please see this report
https://streamable.com/l7m0l3

So default path doesn't work on iOS too, and [OT for this issue] we can't find any valid path that would produce a CSV.

This used to work on v6.00 and now in v6.02 the logging got completely broken. I would guess that this was introduced in https://github.com/vedderb/vesc_tool/pull/299 Logging is a critical tool to provide customer support

Jfriesen222, 2023-04-13:
Oh wow, I will investigate the iOS side today and release a minor fix ASAP. I agree this is core functionality, not sure how I missed that apologies.

nitrousnrg, 2023-04-13:
Thanks!!
The video shows these issues:
1) default path not working
2) no path works, which could be the cause of 1). I'm not sure if the default worked before.
3) scrolling on the directory picker doesn't work
4) you can enable logging on an invalid path

Hope that helps, thanks again!

Jfriesen222, 2023-04-13:
Good point to add a block if the path is no good also

nitrousnrg, 2023-05-05:
> 

@Jfriesen222  I've been asked if this han been fixed, and I don't own any iOS device. Were you able to reproduce this?