vedderb/vesc_tool #316: Old version availability

Karmalakas, 2023-04-28:
It's not an issue per se, but I can't find anywhere older versions of VESC Tool. Website already has v6.02 and I'm looking for v6.00 🙁 Here on GH there are no releases or packages. It would be great if repository had versioned releases with attached compiled apps. I'm looking specifically for Windows and Android versions 🙁 Any chance I could get them somewhere?

Tomblarom, 2023-11-12:
I was looking for exactly this. Found it here: https://forum.esk8.news/t/precompiled-vesc-tool-archive/15920 However it seems like its not intended to have older releases accessible for public domain:

> The matter is a relatively complex one. If a user makes the decision to keep old versions, that is his personal choice. Distributing them is a different matter.
:: https://forum.esk8.news/t/precompiled-vesc-tool-archive/15920/9

 
> Bad news. I received a request from Benjamin to remove binary releases from here:
:: https://forum.esk8.news/t/precompiled-vesc-tool-archive/15920/133

vedderb, 2023-11-13:
The intention is that the latest version can communicate with old firmwares too and to not encourage keeping old versions of vesc tool around. Is there something you can do in the old vesc tool but not in the latest version?

Karmalakas, 2023-11-16:
There was a case, where I was pretty sure, that newer version introduced some issue. So I just wanted to check and confirm. I think I no longer have that issue. Closing this 🙏

closed 2023-11-16 by Karmalakas