vedderb/vesc_tool #322: [Feature Request] Show Motor KV value after motor detection completes

rusins, 2023-06-14:
Motor KV is a useful property to know for end-users, and often the true kv value can differ from what the motor manufacturer stated.

It's an easy thing to calculate (ERPM / motor pole pair count / duty cycle / input voltage) (I think? I'm not sure if 100% duty cycle is linked to input voltage, or the hardware max supported voltage), and could be displayed right under the other motor parameters like resistance and flux linkage after the motor wizzard is run :)

johnrbnsn, 2024-02-11:
Agreed, I would love to see this kV and kT.

TechAUmNu, 2024-03-02:
Is this what you are looking for?
[image]

johnrbnsn, 2024-03-03:
Yes! That would be perfecr!! 

Out of curiosity, can you calculate the kv and kt values from the others listed?

johnrbnsn, 2024-05-28:
@TechAUmNu any updates?